miek/inspectrum #21: Support other file formats

miek, 2015-07-23:
Currently inspectrum supports sample files with complex 32bit floats, it should probably support some other formats:

8bit signed/unsigned ints
16bit signed/unsigned ints ?
real (not complex) samples ?

dominicgs, 2015-08-25:
Ideally I would like to see support for the native formats of most SDR hardware.  Perhaps it would be good to look at the format options supported by other signal analysis software?

garverp, 2015-10-03:
I've done some work on some simple CLI tools for offline data processing (https://github.com/garverp/gr-analysis) and am excited to see this project as it is really needed.  Here are my suggestions for file formats (derivative of the UHD file formats,https://github.com/EttusResearch/uhd/blob/master/host/lib/convert/convert_impl.cpp). I'd love to see inspectrum work with these formats.

Naming convention:
($type)($complex)($size)

$type is the fundamental unit, here is a list:
The list of data types:
Fundamental Data Type,$type=,$size= (bits)
single-precision float,f,32
double-precision float,d,64
integer,i,32
short,s,16
byte,b,8

$complex=c if complex, '' if not complex
$size=size, in bits of fundamental unit, from above

So here are some examples,
fc32 = float fundamental data type, complex, and 32 bits/fundamental data type
sc16 = short fundamental data type, complex, and 16 bits/fundamental data type
f32 = float fundamental data type, not complex, 32 bits/fundamental data type

jboone, 2016-03-18:
See pull request #53 for support of GNU Radio native files, rtl_sdr files, and hackrf_transfer files. I'm hoping it's reasonably easy to extend to additional schemes with the provided SampleAdapter abstraction as a starting point.

miek, 2016-03-18:
Closing this as #53 seems to have covered the major formats. PRs welcome if any common formats are missing.

closed 2016-03-18 by miek