miek/inspectrum #28: Add a tag to the repo

ZeroChaos-, 2015-09-03:
I know for some reason that more and more software developers think that having a public git repo is fine and that releases are unneeded, but for the sake of distros that sometimes want to include your software, having a release (or even just a git tag) is a magical thing.  Looks like things are working fairly well right now, how about a git tag? Seems that 2015.09 is a popular format for tags when you don't want to have to deal with version numbers.

miek, 2015-09-04:
I don't think a git repo alone is fine and I do plan to tag a release soon, but there are a few things I still want to fix up before I'm happy with it.

miek, 2015-10-03:
Done!

closed 2015-10-03 by miek