miek/inspectrum #35: Spectrogram height is limited

miek, 2015-10-03:
The maximum width/height of a QWidget is 16777215 and this is pretty easy to hit with a large file / high zoom / small fft size.
Might need to rethink how the spectrogram drawing is done.

miek, 2016-03-07:
New drawing method since 8a8b800ceeca35f6dcf49e322307eb4de794e314 fixes the widget height issue, but the new issue is the scrollbar value is limited to 32-bits.

sflipper12, 2017-06-06:
Can you please help me i dont know how to launch ispectrum

Thanks
Dale

vsboost, 2017-06-06:
If you did a "make install" you can simply type inspectrum from a terminal window and the program should open.

sflipper12, 2017-06-07:
I used the mac way to install.  It installed all the components but will not launch with just ispectrum

Thank you for your help

Dale

tresacton, 2017-06-14:
what happens when you type the word "inspectrum" in your terminal? can you copy and pasted the text that appears in the terminal window?