miek/inspectrum #42: Example Usage

brannondorsey, 2015-11-03:
Hi there,

I was wondering if there are any examples for how to use inspectrum with data captured from a HackRF or RTL-SDR. I recognize that neither of these Decices produce 32-bit unsigned samples, but is there a way to convert their 8-bit samples for use with inspectrum?

brannondorsey, 2015-11-03:
Correction: HackRF generates 8-bit I and 8-bit Q samples, so I guess it's sample is actually 16-bits.

miek, 2015-11-03:
I like SoX for this sort of thing, this one-liner will convert from HackRF/RTL-SDR 8-bit capture files:
`sox -t s8 in.bin -t f32 out.cfile`

brannondorsey, 2015-11-03:
Awesome, thanks!

closed 2015-11-03 by brannondorsey

brannondorsey, 2015-11-03:
This is such a fantastic tool. Thanks for taking the time to make this :)