miek/inspectrum #47: Time and frequency deltas

proto17, 2015-12-30:
It would be nice to have the ability to do time and frequency deltas like baudline.  Something where you can click to points and get both a high res time and frequency difference between the two selected points.

dnet, 2016-02-27:
Take a look at [my `cursors` branch](https://github.com/dnet/inspectrum/tree/cursors), I implemented a simple cursor mechanism. The current frequency and time of the mouse cursor is displayed in the controls dock, and if you press the mouse key, a two dashed lines become a starting point in both frequency and time, and the differences in time and frequency between this and the mouse cursor is also displayed as delta on the left. Latter is kept even after the mouse button is released, whereas the dashed lines are visible only while the mouse button is held down, marking both cursors.

@proto17 were you thinking about something like this? Please try it out, I'd be interested in what you think. When it's ready, I'll submit it as a pull request.

proto17, 2016-02-28:
That's exactly it!  The only critique I can make is that suppose that I need to measure the time for a signal that takes up more than one screen worth of vertical screen space.  Setting it up so that simply clicking once sets the first marker, and clicking again sets the second marker and shows the delta information.  This might be confusing for some users, so perhaps a check box that allows you to enable single click measurements?  

You did a great job!  Thank you very, very much :D

dnet, 2016-02-28:
Try 649f7f4, I updated the branch. Now there's a checkbox that's checked by default and provides the behavior I originally implemented. But if you uncheck it, the two cursors can be set by two separate clicks and you can scroll between those two. The second click will of course clear both cursors.

proto17, 2016-02-28:
That's perfect!  Thank you so very much!!

dnet, 2016-02-29:
OK, I submitted pull request #48

miek, 2016-03-02:
Fixed by 5ed888fd45998f61fa4ff4e4cf8793ae92a48dec

closed 2016-03-02 by miek