miek/inspectrum #56: This is a question not an issue.

tresacton, 2016-03-23:
I was just wondering whether "Bit Frequency" is effectively synonymous with "baudrate"? I have spent some time researching the relationship between baudrate and symbols etc, however, some of the labels used in Inspectrum appear to me to be specific to Inspectrum, so I wanted to seek clarification on this. 

I am using the waveform branch.

Many thanks!

miek, 2016-03-23:
Yeah, it is synonymous with "baudrate".

I'll probably change that label to "symbol rate" or "symbol frequency" at some point since "bit" wasn't really the right term to use there.

tresacton, 2016-03-31:
Thank you for the confirmation. And also... thank you for the tool itself. You've clearly worked hard on this and it's a time saver for me. Very interested to see how far this evolves.

closed 2016-03-31 by tresacton