miek/inspectrum #57: Crashing of file load.

vsboost, 2016-03-23:
Hi there since the change which included different file formats, i no longer can open up .RAW files that have been generated from Gnuradio --> File Sink.

Program crashes as soon as i Open the file, haven't tested it by renaming the extension to a supported one but thought i would bring it to your attention.

Cheers

miek, 2016-03-23:
Cheers for the report.

For now filenames do need to match the extensions listed in the README. Eventually the file open dialog will allow the user to select the file type and take hints from the extension (without failing), but it'll be a while before that's done.

vsboost, 2016-03-23:
Thanks Miek,

Just for your reference, when i open up the .raw file inspectrum closes with this error
------------------------------------------------------------------------------------------
terminate called after throwing an instance of 'std::runtime_error'
  what():  Invalid file extension
Aborted
--------------------------------------------------------------------------------------------
if i rename .raw to .cfile all is good.

Thanks for making this very useful program.

Tom

jboone, 2016-03-23:
I'm afraid that's my code. I didn't know what the policy and behavior should be around extensions not explicitly supported, so I throw an exception. Either it needs to get caught and handled more gracefully, or better, we eventually do what miek suggested up-thread. Perhaps my short-term decision to throw an exception was a little harsh. ;-)

vsboost, 2016-03-23:
No probs Jared, i really should use .cfile in my file sinks, just an old habit using .raw

miek, 2016-04-24:
Defaults to float as of 9b774f0845ceafa757e6ec79d6c9a4a7d0d6d166

closed 2016-04-24 by miek