miek/inspectrum #60: Build fails on OS X, seems to be related to Qt5

daveio, 2016-04-02:
cmake succeeds (after also installing `gnuradio` from Homebrew), but make fails. I'm afraid I have little to no familiarity with Qt, so I'm hoping the error vomit makes sense to you. Any assistance appreciated - let me know if you want any more details or want me to try something.

Dependencies:

```
 dave  ~  for i in qt5 fftw cmake pkg-config gnuradio; do brew info $i | egrep -B1 Poured; done
/usr/local/Cellar/qt5/5.6.0 (7,288 files, 310.9M)
  Poured from bottle
/usr/local/Cellar/fftw/3.3.4_1 (45 files, 10.2M) *
  Poured from bottle
/usr/local/Cellar/cmake/3.5.1 (2,010 files, 27.5M) *
  Poured from bottle
/usr/local/Cellar/pkg-config/0.29.1 (10 files, 627.2K) *
  Poured from bottle
/usr/local/Cellar/gnuradio/3.7.9.1 (3,350 files, 125.4M) *
  Poured from bottle
```
cmake:

```
 dave  ~  src  inspectrum  build  CMAKE_PREFIX_PATH=/usr/local/opt/qt5/lib/cmake cmake ..
-- The CXX compiler identification is AppleClang 7.3.0.7030029
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PkgConfig: /usr/local/bin/pkg-config (found version "0.29.1")
-- Boost version: 1.60.0
-- Found the following Boost libraries:
--   system
--   program_options
Checking for GNU Radio Module: RUNTIME
-- Checking for module 'gnuradio-runtime'
--   Found gnuradio-runtime, version 3.7.9.1
 * INCLUDES=/usr/local/Cellar/gnuradio/3.7.9.1/include
 * LIBS=/usr/local/Cellar/gnuradio/3.7.9.1/lib/libgnuradio-runtime.dylib;/usr/local/Cellar/gnuradio/3.7.9.1/lib/libgnuradio-pmt.dylib
-- Found GNURADIO_RUNTIME: /usr/local/Cellar/gnuradio/3.7.9.1/lib/libgnuradio-runtime.dylib;/usr/local/Cellar/gnuradio/3.7.9.1/lib/libgnuradio-pmt.dylib
GNURADIO_RUNTIME_FOUND = TRUE
Checking for GNU Radio Module: ANALOG
-- Checking for module 'gnuradio-analog'
--   Found gnuradio-analog, version 3.7.9.1
 * INCLUDES=/usr/local/Cellar/gnuradio/3.7.9.1/include
 * LIBS=/usr/local/Cellar/gnuradio/3.7.9.1/lib/libgnuradio-analog.dylib;/usr/local/Cellar/gnuradio/3.7.9.1/lib/libgnuradio-runtime.dylib;/usr/local/Cellar/gnuradio/3.7.9.1/lib/libgnuradio-pmt.dylib;/usr/local/Cellar/gnuradio/3.7.9.1/lib/libvolk.dylib
-- Found GNURADIO_ANALOG: /usr/local/Cellar/gnuradio/3.7.9.1/lib/libgnuradio-analog.dylib;/usr/local/Cellar/gnuradio/3.7.9.1/lib/libgnuradio-runtime.dylib;/usr/local/Cellar/gnuradio/3.7.9.1/lib/libgnuradio-pmt.dylib;/usr/local/Cellar/gnuradio/3.7.9.1/lib/libvolk.dylib
GNURADIO_ANALOG_FOUND = TRUE
Checking for GNU Radio Module: BLOCKS
-- Checking for module 'gnuradio-blocks'
--   Found gnuradio-blocks, version 3.7.9.1
 * INCLUDES=/usr/local/Cellar/gnuradio/3.7.9.1/include
 * LIBS=/usr/local/Cellar/gnuradio/3.7.9.1/lib/libgnuradio-blocks.dylib;/usr/local/Cellar/gnuradio/3.7.9.1/lib/libgnuradio-runtime.dylib;/usr/local/Cellar/gnuradio/3.7.9.1/lib/libgnuradio-pmt.dylib
-- Found GNURADIO_BLOCKS: /usr/local/Cellar/gnuradio/3.7.9.1/lib/libgnuradio-blocks.dylib;/usr/local/Cellar/gnuradio/3.7.9.1/lib/libgnuradio-runtime.dylib;/usr/local/Cellar/gnuradio/3.7.9.1/lib/libgnuradio-pmt.dylib
GNURADIO_BLOCKS_FOUND = TRUE
Checking for GNU Radio Module: FILTER
-- Checking for module 'gnuradio-filter'
--   Found gnuradio-filter, version 3.7.9.1
 * INCLUDES=/usr/local/Cellar/gnuradio/3.7.9.1/include
 * LIBS=/usr/local/Cellar/gnuradio/3.7.9.1/lib/libgnuradio-filter.dylib;/usr/local/Cellar/gnuradio/3.7.9.1/lib/libgnuradio-fft.dylib;/usr/local/Cellar/gnuradio/3.7.9.1/lib/libgnuradio-runtime.dylib;/usr/local/Cellar/gnuradio/3.7.9.1/lib/libgnuradio-pmt.dylib
-- Found GNURADIO_FILTER: /usr/local/Cellar/gnuradio/3.7.9.1/lib/libgnuradio-filter.dylib;/usr/local/Cellar/gnuradio/3.7.9.1/lib/libgnuradio-fft.dylib;/usr/local/Cellar/gnuradio/3.7.9.1/lib/libgnuradio-runtime.dylib;/usr/local/Cellar/gnuradio/3.7.9.1/lib/libgnuradio-pmt.dylib
GNURADIO_FILTER_FOUND = TRUE
-- Checking for module 'fftw3f'
--   Found fftw3f, version 3.3.4
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/dave/src/inspectrum/build
```

make:

[... 79 lines cut ...]
template<typename T>
                  ^
/usr/local/opt/qt5/lib/QtCore.framework/Headers/qdebug.h:284:60: error: no member named 'Value' in the global namespace
typename QtPrivate::QEnableIf<QtPrivate::IsQEnumHelper<T>::Value, QDebug>::Type
                                                         ~~^
/usr/local/opt/qt5/lib/QtCore.framework/Headers/qdebug.h:284:65: error: a template declaration can only declare a single entity
typename QtPrivate::QEnableIf<QtPrivate::IsQEnumHelper<T>::Value, QDebug>::Type
                                                                ^
/usr/local/opt/qt5/lib/QtCore.framework/Headers/qdebug.h:375:1: error: unknown type name 'Q_FORWARD_DECLARE_OBJC_CLASS'
Q_FORWARD_DECLARE_OBJC_CLASS(NSObject);
^
/usr/local/opt/qt5/lib/QtCore.framework/Headers/qdebug.h:376:34: error: unknown type name 'Q_FORWARD_DECLARE_CF_TYPE'
QT_FOR_EACH_CORE_FOUNDATION_TYPE(QT_FORWARD_DECLARE_CF_TYPE)
                                 ^
/usr/local/opt/qt5/lib/QtCore.framework/Headers/qdebug.h:376:34: error: unknown type name 'Q_FORWARD_DECLARE_CF_TYPE'
/usr/local/opt/qt5/lib/QtCore.framework/Headers/qdebug.h:376:34: error: unknown type name 'Q_FORWARD_DECLARE_CF_TYPE'
/usr/local/opt/qt5/lib/QtCore.framework/Headers/qdebug.h:376:34: error: unknown type name 'Q_FORWARD_DECLARE_CF_TYPE'
/usr/local/opt/qt5/lib/QtCore.framework/Headers/qdebug.h:376:34: error: unknown type name 'Q_FORWARD_DECLARE_CF_TYPE'
/usr/local/opt/qt5/lib/QtCore.framework/Headers/qdebug.h:376:34: error: unknown type name 'Q_FORWARD_DECLARE_CF_TYPE'
fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.
make[2]: *** [CMakeFiles/inspectrum.dir/cursors.cpp.o] Error 1
make[1]: *** [CMakeFiles/inspectrum.dir/all] Error 2
make: *** [all] Error 2
```

daveio, 2016-04-02:
I noticed qt5 is keg-only - I have managed to build by doing

```
brew unlink qt
brew link --force qt5
cmake -DCMAKE_PREFIX_PATH:STRING=/usr/local/opt/qt5/lib/cmake/ ..
make install
brew unlink qt5
brew link qt
```

Might be worth massaging the make system to avoid having to force-link qt5, though.

saerdnaer, 2016-05-15:
Had the same issue on OSX 10.11.4. Thanks for your solution!

saerdnaer, 2016-09-11:
The solution from issue #40 looks litte better
```
CMAKE_PREFIX_PATH=$(brew --prefix qt5)/lib/cmake cmake .. -DCMAKE_CXX_FLAGS:STRING=-I/usr/local/opt/qt5/include ..
make install
```

miek, 2016-09-12:
Fixed in #87

closed 2016-09-12 by miek