miek/inspectrum #62: Feature request for adding notes

vsboost, 2016-04-19:
Hi Miek,

Sorry one more feature request if possible, i would love to have the ability to add some notes to a file, in particular the 1's and 0's lined up with the cursors and the ability to copy and paste them to the clipboard, something like what i have attached.  Thanks again.
[image]

miek, 2016-04-19:
So there's a plan for something like this that I'm working towards in the waveform branch, but it's not going to be notes that you can save/load/copy.

The idea is to display extra plots with simple demod functions (frequency, amplitude, etc.) then use the timing info. from the cursors to extract symbols automatically from the thresholded plot (bottom one in pic):
[image]

I've made #63 to keep track of that so I'll close this. Cheers!

closed 2016-04-19 by miek

vsboost, 2016-04-19:
That seems an even better idea, thanks again.

[inspectrum-notes]<https://cloud.githubusercontent.com/assets/578095/14658660/8e625c78-068d-11e6-87a2-38e218107cf0.jpg>