miek/inspectrum #66: building waveform

vsboost, 2016-04-23:
Hi miek,

Im not sure what i have done here but i just tried to build the latest waveform commits and my inspectrum looks like the master branch have lost the horizontal selection bars.  I wanted to see the plotview: Basic symbol extraction work you just made.
 
even started from scratch

git clone https://github.com/miek/inspectrum.git
git checkout waveform
mkdir build
cmake ..
make
./inspectrum

Any suggestions

miek, 2016-04-24:
The tuner overlay only shows up when there's a derived plot using it (so it doesn't get in the way if you only want to look at the spectrogram).
You can right click on the spectrogram to add a plot.

This feature won't be that useful for the signal you're looking at, since there's no amplitude plot yet.

vsboost, 2016-04-24:
Ahh thanks miek,

closed 2016-04-24 by vsboost

miek, 2016-04-24:
Added an amplitude plot and scaling, enjoy :)

[image]

Note: the threshold plot won't work well on this (need to think about it some more), but you can paste into python or whatever to threshold it easily.

vsboost, 2016-04-24:
Thanks miek,
Looking good :-)
[image]

vsboost, 2016-04-24:
Loving this, and thanks for the python script to convert to 1's & 0's.  Ill just add it to this comment for reference.

syms = [1.99554, 0.00911547, 2.04077, 0.0130324, 2.02553, 0.0156159, 2.00305, 0.00284353, 1.92315, 0.0208421, 1.9628, 0.00791253, 1.96002, 0.0145462, 1.93397, 0.0201539, 1.95594, 0.0139064, 1.93589]
print map(lambda x: 1 if x > 1 else 0, syms)