miek/inspectrum #69: Enhancement remove dependency to Gnu Radio

bvernoux, 2016-05-02:
Does it is planned to remove the dependency to gnuradio 3.7.x ?
* It will be great for easier rebuild on different platform (especially Windows)

miek, 2016-05-10:
I'm not sure, but I have been thinking about it.

I don't really know what to do instead. [liquid-dsp](http://liquidsdr.org/doc/) looks nice but it doesn't seem to be packaged for any distros. I could also just do all the DSP from scratch, the only complicated thing in there is FIR design. Any other ideas?

garverp, 2016-05-10:
Why rewrite all the core DSP operations? That's a huge task, I think. The GNURadio libraries have a large community and therefore get a lot of attention and scrutiny. There's also the performance optimization work using VOLK to leverage the SSE/AVX/SIMD instructions on general purpose processors. 

There are certainly other good libraries out there besides GNURadio, like you mentioned liquid-dsp. My suggestion is that Inspectrum abstract the details in the form of an API to do DSP things and let folks write plug-ins for whatever DSP library they want. You'll want that decoupling.

From a software architecture point of view, I think it's a mistake to rewrite all DSP operations. Then you only have folks using Inspectrum debugging and fixing instead of a mature GR community or whatever other library you want. I don't think the resources exist to do it well and that's not what I see as really lacking in the software radio community. I think it's the visualization, manipulation, and analysis of offline data. Filters, demods, bitsync etc, work well in GR. It's the integration into a visualization tool which is the value-added to the community.

PWG

miek, 2016-05-10:
To expand a bit more, these are the current downsides of using GNURadio as I see it:
 * It's a big dependency - it pulls in a huge amount of stuff compared with how much I actually use from it. https://travis-ci.org/dominicgs/inspectrum/jobs/125588282#L700
 * It's designed for working with continuous streams, whereas I'm processing small chunks of non-contiguous data and it feels like I'm shoehorning it into a use-case it's not meant for.

The DSP I use from GNURadio is mostly trivial, so that's why I see doing it from scratch as an option. Amplitude is the magnitude of a vector and instantaneous frequency is the angle between two vectors, but FIR design is a bit more complicated. Using just VOLK could be a good option too.

I don't aim to cover every possible use-case, so I'm not sure whether the DSP is going to get much more complex. I see this tool as a way to fully deal with simple signals, or an easier way to get enough information about a recording to then write a GNURadio flowgraph to go further.

schneider42, 2016-05-10:
VOLK is a great library to just embed in your own project. If you look at the basic FIR filter implementation in GNURadio, you will see that it does not take a lot of effort to implement a filter using VOLK.

Designing the filter taps is something different though. I guess other DSP libraries do provided decent tools for that.

garverp, 2016-05-11:
Thanks for the discussion. 

I can see your point if you are only planning a few DSP operations it may not make sense. You could still abstract it to allow for various DSP library implementations, though. E.g. you could code a default implementation in VOLK and then if GNURadio/liquid-dsp/etc is available it could be compiled with those libraries. VOLK should be able to do magnitudes, angle between vectors, basic vector/matrix operations. Also, if you abstracted it would be easy to compile with Intel MKL or other math libraries. This probably makes more sense for more complicated tasks such as filter design, etc.

As for two, have you seen Tim O'Shea's work on [gr-eventstream](https://github.com/osh/gr-eventstream)? It allows GR to handle finite-length "events", which I believe is what you are after. Some of the code is being integrated into mainline GR.

I've mentioned this before, but I think a great integration between GR and Inspectrum would be inspectrum passes an async message a la gr-eventstream to GR, GR processes, and then sends a message back. Then, Inspectrum displays the results. So by updating/tweaking a flowgraph you can get immediate visual feedback.

PWG

miek, 2016-06-11:
I changed my mind a bit and have been working on a liquid-dsp port: https://github.com/miek/inspectrum/tree/liquid - seems to work well but I'd appreciate any testing, it depends on git master of liquid-dsp.

miek, 2016-07-03:
Port to liquid-dsp has been merged

closed 2016-07-03 by miek