miek/inspectrum #72: build fails

saerdnaer, 2016-05-16:
gnuradio@vbox:~/inspectrum/build$ make
[  4%] Automoc for target inspectrum
[  4%] Built target inspectrum_automoc
Linking CXX executable inspectrum
/usr/bin/ld: CMakeFiles/inspectrum.dir/rotator_ex_cc.cpp.o: undefined reference to symbol 'volk_32fc_s32fc_x2_rotator_32fc'
/usr/local/lib/libvolk.so.1.2.2: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [inspectrum] Error 1
make[1]: *** [CMakeFiles/inspectrum.dir/all] Error 2
make: *** [all] Error 2

nickma82, 2016-05-21:
Hi,
same here on debian.

After installing the dependencies:
apt-get install libboost-dev libboost-program-options-dev libboost-system-dev liblog4cpp5-dev libvolk-dev

### Output was:
/usr/bin/ld: CMakeFiles/inspectrum.dir/rotator_ex_cc.cpp.o: undefined reference to symbol 'volk_32fc_s32fc_x2_rotator_32fc'
//usr/lib/x86_64-linux-gnu/libvolk.so.0.0.0: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
CMakeFiles/inspectrum.dir/build.make:572: recipe for target 'inspectrum' failed
make[2]: *** [inspectrum] Error 1
CMakeFiles/Makefile2:60: recipe for target 'CMakeFiles/inspectrum.dir/all' failed
make[1]: *** [CMakeFiles/inspectrum.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

### Solution
Add [libvolk](http://libvolk.org/) as to be linked, e.g. by adding `volk` to the CMakeList.txt `target_link_libraries` (See [the pastebin link](http://pastebin.com/NnBuKH4U))

.

miek, 2016-07-03:
This issue should be gone in master now since the move from GNURadio -> liquid-dsp

closed 2016-07-03 by miek