miek/inspectrum #73: Question: Vertical/horizontal

theforcedk, 2016-05-29:
Hi there

All the tutorials/guides display the data vertical, but when I import my file from hackrf_transfer (hackrf_transfer -r 27165000.s8 -g 16 -l 32 -f 27165000), and converted via. sox (sox -t s8 27165000.s8 -t f32 out.cfile) data is being displayed horizontally - is this intentional/why does it do that? :)

[image]

miek, 2016-06-01:
It's a fairly recent change: the current stable release has time on the vertical axis, but the git master has time on the horizontal axis.
I changed it mainly because I'm adding additional time-domain plots that I want to display alongside the spectrogram, and it makes a lot more sense to display them with time on the horizontal axis. Also I think it fits better on a wide-screen which most people have.

theforcedk, 2016-06-01:
@miek Great; I like it better this way too. Was just confused as most of the guides was displaying vertically :) thanks

closed 2016-06-01 by theforcedk