miek/inspectrum #81: Desktop file missing

drws, 2016-08-25:
The .desktop file providing a method to launch inspectrum in Linux DEs is missing and the only way to run inspectrum is from the command line. This is working fine, but as the application is GUI-based, I think the missing .desktop file is a bug (and the missing icon image is a low priority enhancement). The sample content of a working such file that I use would be:

    [Desktop Entry]
    Version=1.0
    Type=Application
    Name=inspectrum
    Exec=inspectrum
    Icon=
    Categories=HamRadio;

drws, 2021-07-06:
After almost 5 years I dare to bump this :)

schneider42, 2021-07-12:
@miek I guess inspectrum is in need of a logo/icon ;)

drws, 2021-07-12:
@schneider42 An icon would be a welcome addition, but it is [not a requirement](https://specifications.freedesktop.org/desktop-entry-spec/desktop-entry-spec-1.1.html) for a working .desktop file.