miek/inspectrum #88: Create some UI for symbol extraction results

miek, 2016-09-15:


nihalpasham, 2018-09-22:
This would be great. I seem to be getting inconsistent results when I try to extract symbols from an FSK modulated-Manchester encoded signal. The usual low-0 or high-1 works well though.

konsgn, 2022-01-20:
Dropping this here since I stumbled upon this post while thinking on how to figure this out:

Set sample rate at 2x the symbol rate, then just decode every bit pair and look for 10 or 01.

```
#!/bin/python
import sys

def DangerousGame(bit):
  bits = []
  for i in range(int(len(bit)/2)):
    notpacked = 1
    if bit[i*2:i*2+2] == [1,0]:
      bits.append(1)
      notpacked = 0
    else:
      if bit[i*2:i*2+2] == [0,1]:
        bits.append(0)
        notpacked = 0
    if notpacked:
      print("massive Error at %d"%i)
  return bits
```