miek/inspectrum #90: Inspectrum requires file to open

erichahn, 2016-09-16:
I tried to open inspectrum after a fresh install and it immediately opens a window that grabs the screen behind this new window that opens. Next the window freezes, grays out, and never opens.

I next tried opening inspectrum and also gave it a file to look at and it finally worked!

I don't think inspectrum should REQUIRE a file to open properly. I believe this is a bug, unless it is a design choice (that I don't agree with, but I'm only just starting to use it).

Any thoughts?

[image]

miek, 2016-09-16:
That's a bug, looks like I broke it in the last few commits and caused an inf. loop in `PlotView::paintTimeScale`. I'll get it fixed soon, cheers for the report!

schneider42, 2016-09-16:
I guess `durationPerTick` becomes zero and the loops never finish. Both loops should get a check if the used step is > 0.

miek, 2016-09-17:
Fixed in d747e9daffbde3e45df56bb2804e2d93fa6eaa54

closed 2016-09-17 by miek