miek/inspectrum #91: Fix - Installation Guide - Building on Debian-based distros

caiotelles, 2016-09-23:
I would like to propose the below adjustments to the installation guide.

## **Building on Debian-based distros**
```
git clone git://github.com/jgaeddert/liquid-dsp.git
cd liquid-dsp
./bootstrap.sh
./configure
make
sudo make install
cd ..
sudo apt-get install qt5-default libfftw3-dev cmake pkg-config
mkdir build
cd build
cmake ..
cd ..
make
sudo make install
```

Tks

ttmatis, 2017-02-22:
Thanks for for this, caiotelles.    Much appreciated!    Please note, the 3rd to last line 'cd ..'  needs to be removed (at least, in my case).    Again, many thanks.

tomevans80, 2019-08-11:
I just had the same problem with Cmakelists being missing, as I found this page before the main Inspectrum instructions. These need to be read alongside those. So the first lines in the instructions should actually be:

https://github.com/miek/inspectrum.git
cd inspectrum

Then it works.

closed 2023-10-22 by miek