miek/inspectrum #108: Enable cursors forgets positioning

samyk, 2016-12-19:
Suggestion / minor issue: When enabling "Enable cursors" and setting everything nicely, then unchecking "Enable cursors" (to get a better view for example), and then re-enabling it, the settings (except for Symbols) are erased, forcing the user to restretch the cursor pane.

miek, 2016-12-19:
This was done on purpose for the case where you scroll away from where the cursors were, you can just toggle them off and on to get them back in a sane position. But I'm open to any better way of doing it.

samyk, 2016-12-19:
I see. I'd prefer it kept its location and instead offered a 'Reset' or similar button next to the checkbox to get them in a sane position.

Additionally, when disabling the 'Enable cursors' checkbox, I would still like to see the Rate/Period/Symbol Rate/Symbol Periods rather than hiding them, as sometimes I just want a cleaner view of the signal.