miek/inspectrum #111: Convert controls into a top/bottom docked toolbar

samyk, 2016-12-19:
Suggestion: The controls take a lot of screen real estate, at least on a typical laptop screen/resolution. I like that they're detachable but that's a bit clumsy to use. I would love if the controls were in a thin toolbar at the bottom, and the time selection text (rate/period/etc) could just overlay on the signal itself. Happy to make a screenshot of what sort of thing I would like.

I'll start learning some Qt soon so instead of throwing a ton of suggestions I can actually try and help!

miek, 2016-12-19:
Completely agree, it's been on the todo list in my head I tend to get stuck on how it should look. If you want to do some mockups that would be great.

fsch2, 2017-01-23:
I think this is a great idea. Also, you will get quite far with the usual icons, combo boxes and spin boxes.

[image]

777arc, 2021-07-02:
So, in Ubuntu 20, the controls doesn't show up at all, I didn't realize it existed until I starting playing around with keys to see what hotkeys existed, and when I hit the menu key (next to right side Control) I saw that pop-up about adding plots and extracting symbols and it blew my mind =P.  So yeah it should probably be more obvious, I would have never known it existed.  Maybe just add it as a normal button in the GUI?