miek/inspectrum #113: Support .wav files

samyk, 2016-12-19:
Feature request: It would be nice to be able to open .wav files straight into inspectrum.

itdaniher, 2017-02-04:
This gains additional value in the context of migrating from other tools, ie [Baudline](http://www.baudline.com/), which natively exports `RIFF (little-endian) data, WAVE audio, Microsoft PCM, 16 bit, stereo`

itdaniher, 2017-02-05:
```
import scipy.io.wavfile
import sys

filename = sys.argv[1]

fs1, y1 = scipy.io.wavfile.read(filename)

assert y1.dtype.name == 'int16'

stereo_to_complex = lambda a: a[0] + a[1]*1j
y1 = y1/65536.

y1 = stereo_to_complex(y1.T).astype('complex64')

filename = filename.replace('wav', 'cf32')

y1.tofile(str(fs1)+'.'+filename)
```

boring stack-overflow-derived solution that worked for me!

reald, 2020-11-13:
Created pull request  https://github.com/miek/inspectrum/pull/176 . Alternatively you can set the sampling depth manually:

`inspectrum -f cs16 file.wav`