miek/inspectrum #119: debian build errors

Gary201611, 2017-01-08:
Hi,
miek,I have some questions about build inspectrum on Debain,below is the mistakes

root@debian:/home/gary/inspectrum/build# make
Scanning dependencies of target inspectrum_automoc
[  4%] Automatic moc for target inspectrum
Generating moc_cursor.cpp
Generating moc_cursors.cpp
Generating moc_mainwindow.cpp
Generating moc_plot.cpp
Generating moc_plotview.cpp
Generating moc_spectrogramcontrols.cpp
Generating moc_spectrogramplot.cpp
Generating moc_traceplot.cpp
Generating moc_tuner.cpp
[  4%] Built target inspectrum_automoc
Scanning dependencies of target inspectrum
[  9%] Building CXX object CMakeFiles/inspectrum.dir/amplitudedemod.cpp.o
[ 13%] Building CXX object CMakeFiles/inspectrum.dir/cursor.cpp.o
[ 18%] Building CXX object CMakeFiles/inspectrum.dir/cursors.cpp.o
[ 22%] Building CXX object CMakeFiles/inspectrum.dir/main.cpp.o
[ 27%] Building CXX object CMakeFiles/inspectrum.dir/fft.cpp.o
[ 31%] Building CXX object CMakeFiles/inspectrum.dir/frequencydemod.cpp.o
/home/gary/inspectrum/frequencydemod.cpp: In member function ‘virtual void FrequencyDemod::work(void*, void*, int, off_t)’:
/home/gary/inspectrum/frequencydemod.cpp:33:5: error: ‘freqdem’ was not declared in this scope
     freqdem fdem = freqdem_create(0.05f);
     ^
/home/gary/inspectrum/frequencydemod.cpp:35:28: error: ‘fdem’ was not declared in this scope
         freqdem_demodulate(fdem, in[i], &out[i]);
                            ^
/home/gary/inspectrum/frequencydemod.cpp:35:48: error: ‘freqdem_demodulate’ was not declared in this scope
         freqdem_demodulate(fdem, in[i], &out[i]);
                                                ^
/home/gary/inspectrum/frequencydemod.cpp:37:21: error: ‘fdem’ was not declared in this scope
     freqdem_destroy(fdem);
                     ^
/home/gary/inspectrum/frequencydemod.cpp:37:25: error: ‘freqdem_destroy’ was not declared in this scope
     freqdem_destroy(fdem);
                         ^
CMakeFiles/inspectrum.dir/build.make:169: recipe for target 'CMakeFiles/inspectrum.dir/frequencydemod.cpp.o' failed
make[2]: *** [CMakeFiles/inspectrum.dir/frequencydemod.cpp.o] Error 1
CMakeFiles/Makefile2:60: recipe for target 'CMakeFiles/inspectrum.dir/all' failed
make[1]: *** [CMakeFiles/inspectrum.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

Hope you help me,looking forward your reply!

Gary201611, 2017-01-08:
By the way,My computer system is Debian GNU/Linux 8 (jessie) 64-bit.
All of the dependency library includ liquid-dsp is installed.

miek, 2017-04-17:
If this is still an issue, it was probably an old version of liquid-dsp. Feel free  to re-open if you're still having issues.

closed 2017-04-17 by miek