miek/inspectrum #125: Add FFT overlap slider

mhostetter, 2017-03-29:
This is a feature request.

It would be nice to add an FFT overlap feature. Maybe beneath the "FFT Size" slider you could add an "FFT Overlap" slider. I would suggest a range `0`, `2`, `4`, ..., `32`, `64` in powers of 2, where `0` is no overlap, `2` is 2 overlaps (shift by 1/2 the FFT size), `4` is 4 overlaps (shift by 1/4 the FFT size), etc. The default could be `0`.

I would also suggest displaying the number beside the slider. I would also suggest adding the number beside the "FFT Size" slider as well.

miek, 2017-03-29:
If I understand you correctly, this is what the "zoom" slider does already.

mhostetter, 2017-03-29:
Oh, ok. I guess I didn't realize that's what it did. Thanks.

I do think having the numbers beside the sliders would help the user, but maybe that's for another issue.

closed 2017-03-29 by mhostetter