miek/inspectrum #142: Cannot find xcb?

mike65535, 2018-02-06:
$ inspectrum

This application failed to start because it could not find or load the Qt platform plugin "xcb"
in "".

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, webgl, xcb.

Reinstalling the application may fix this problem.
Aborted (core dumped)

miek, 2018-02-06:
Some googling around suggests this might be caused by mismatched Qt libraries.

Could you share some more details about your setup:

  * OS & version/distro
  * How you installed Qt
  * How you installed inspectrum (built from source?)

mike65535, 2018-02-06:
Fedora 27
I ran   dnf install qt5-default
I built inspectrum

I had issues with the inspectrum build not finding the qt stuff.  Lots of googling had me trying this and that to get the build to find Qt5Widgets and Qt5Concurrent. Eventuall the  build worked but running inspectrum throws that error I reported.

mike65535, 2018-02-12:
>>How you installed inspectrum (built from source?)

I reported that I built from source - is there any other way?

miek, 2018-02-15:
There are some distro packages around, but for most people building from source is still recommended.

As for the error, I've done some searching around but I can't see a clear cause or how to fix it.
You might get some more info by running with `QT_DEBUG_PLUGINS=1` set.

mike65535, 2018-02-15:
OK, ran with `QT_DEBUG_PLUGINS=1` and got this:
```
Got keys from plugin meta data ("xcb")
QFactoryLoader::QFactoryLoader() checking directory path "/usr/local/bin/platforms" ...
Cannot load library /usr/local/Qt-5.10.1/plugins/platforms/libqxcb.so: (/usr/lib64/libQt5XcbQpa.so.5: symbol _ZN21QPlatformBackingStore15composeAndFlushEP7QWindowRK7QRegionRK6QPointP20QPlatformTextureListP14QOpenGLContextb, version Qt_5_PRIVATE_API not defined in file libQt5Gui.so.5 with link time reference)
QLibraryPrivate::loadPlugin failed on "/usr/local/Qt-5.10.1/plugins/platforms/libqxcb.so" : "Cannot load library /usr/local/Qt-5.10.1/plugins/platforms/libqxcb.so: (/usr/lib64/libQt5XcbQpa.so.5: symbol _ZN21QPlatformBackingStore15composeAndFlushEP7QWindowRK7QRegionRK6QPointP20QPlatformTextureListP14QOpenGLContextb, version Qt_5_PRIVATE_API not defined in file libQt5Gui.so.5 with link time reference)"
This application failed to start because it could not find or load the Qt platform plugin "xcb"
in "".

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, webgl, xcb.

Reinstalling the application may fix this problem.
Aborted (core dumped)
```

miek, 2018-02-15:
I think you may have a mix of Qt versions installed. As far as I can see, the latest version available for Fedora 27 is 5.9.4, but you have references to 5.10.1 in the errors.
I also can't find a package named `qt5-default`, so I'm not entirely sure what you've got installed.

I suggest clearing out all Qt stuff and reinstalling from your package manager.

mike65535, 2018-02-16:
I beat the crap out of qt (I now have 5.9.4 "installed") and not sure if I've improved things, but now I get a different error:

`inspectrum: /usr/lib64/libQt5Core.so.5: version `Qt_5.10' not found (required by inspectrum)
`
So why is inspectrum insisting on 5.10?

miek, 2018-02-16:
OK great, glad you got it running in the end.

I don't have any experience with Fedora so I don't know the right way over there. I would expect that you can install Qt with dnf and everything just work.
inspectrum may still have been looking for old files if it wasn't completely rebuilt after reinstalling Qt.
Manually copying libs around is rarely the right solution and can come back to bite later when versions end up mismatched, so beware.

closed 2018-02-16 by miek