miek/inspectrum #144: Building on Ubuntu 16.04 Xenial - dpkg commands incomplete??

wmweis62, 2018-02-15:
I am trying to install inspectrum on Lubuntu 16.04 and I am using the instructions for Ubuntu 16.04 Xenial.  Are the two dpkg commands incomplete? Are they missing directory information? I get errors if follow the documented guideline for installing inspectrum on Ubuntu.

dpkg -x libliquid1d_1.3.0-1_amd64.deb 
dpkg -x libliquid-dev_1.3.0-1_amd64.deb

miek, 2018-02-15:
Many of the instructions on the Build wiki page are contributed by other users, so I've not tested them all myself. If you're able to figure out the correct commands please do update the page.

If you're not able to get up and running with those instructions. I suggest building [liquid-dsp](https://github.com/jgaeddert/liquid-dsp) from source, and then following the generic 'Debian-based' install instructions (without apt installing libliquid-dev). This is what I do on Ubuntu 16.04.

wmweis62, 2018-02-15:
Thanks so much for your reply Mike!  I am currently watching your youtube video.

I was able to get inspectrum installed by using pybombs.

closed 2018-02-15 by miek