miek/inspectrum #146: Build fails with Qt 5.11.0

apjanke, 2018-05-31:
Hi folks,

The build is failing for me on OS X 10.13.4 with the new Qt 5.11.0 installed.

Fails during cmake with the following error:

```
-- Found FFTW: /usr/local/lib/libfftw3f.dylib  
-- Found LIQUID: /usr/local/lib/libliquid.dylib  
CMake Error at CMakeLists.txt:72 (qt5_use_modules):
  Unknown CMake command "qt5_use_modules".

-- Configuring incomplete, errors occurred!
See also "/tmp/inspectrum-20180523-36493-1vl3exo/inspectrum-0.2.1/CMakeFiles/CMakeOutput.log".

```

Example build logs: https://gist.github.com/ilovezfs/b0955c7da0cbb43f553ae58ff03eba85

Looks like the same issue as this: https://github.com/lastfm/liblastfm/pull/36

apjanke, 2018-06-01:
Thanks!

Still broken for me locally, I'm afraid. https://github.com/miek/inspectrum/pull/147#issuecomment-393998218  Can't tell if it's a local configuration issue on my end or an actual issue in inspectrum. Will look in to it if I have time over the weekend.

miek, 2018-06-06:
So I've merged that change as it fixed my builds anyway. I'll keep this open to track any further changes needed for Homebrew & a new release.

miek, 2018-06-07:
Great, thanks! I've just tagged v0.2.2

apjanke, 2018-06-07:
Homebrew is updated: https://github.com/Homebrew/homebrew-core/pull/28734

closed 2018-06-07 by apjanke