miek/inspectrum #152: Help with .cs16 file format - please

NoisySignal, 2018-08-13:
Does anyone know where I might find the .cs16 file specification?

Some web reports mention a header that contains things like the sample rate and the centre frequency. One note suggests that the header length is 24 bytes. If I know the elements and their format then I could read this header correctly.

I am not sure if inspectrum is reading in my .cs16 files correctly.

I'd appreciate any help in sorting this out.

mike65535, 2018-08-13:
CS16 files do not have a header AFAIK.  Inspectrum would not have you, the
user, enter a sample rate if the files had that info.

I believe the files are just IQ pairs of 16-bit integers.

-Mike

jboone, 2018-08-15:
The nomenclature breaks down like this:

"c": complex (pairs of numbers per sample).
"s", "u", or "f": signed integer, unsigned integer, or floating point.
"8", "16", or "32": for the size in bits of the numbers composing the sample.

The most common sample type in GNU Radio is ".cf32": complex numbers composed of two 32-bit floating point numbers. People use ".cfile" extensions for this as well, and I believe Inspectrum recognizes that.

The type of a HackRF sample stream is ".cs8": complex numbers composed of two 8-bit signed integers. On the PortaPack, there's enough processing gain that files saved to the SD card are ".cs16" to preserve that little bit of extra dynamic range.

Not sure what other SDRs use, but I believe the RTL-SDRs are ".cu8" (complex numbers composed of unsigned 8-bit integers). I assume USRPs, BladeRFs, LimeSDRs, etc. would use ".cs16" (complex numbers composed of signed 16-bit integers). But I could be way off-base on that...

jboone, 2018-08-15:
And to actually answer the question... None of these file types have a header. Sorry. I quickly lost track of the point of your question, as I often do.

NoisySignal, 2018-08-17:
Thanks for your help @mike65535 and @jboone !

For your information, it was the post by f4exb in the following link that suggested that .cs16 files had a header:
  https://github.com/f4exb/sdrangel/issues/10

Because I was new to the format it was hard for me to know what information was true (I didn't have access to the code that generate the .cs16 files).

Thanks again!

closed 2018-08-17 by NoisySignal