miek/inspectrum #158: inspectrum hangs when specifying high sample rates

miek, 2019-09-04:
To reproduce: `inspectrum --rate 10e9`

It seems to be stuck in `paintFrequencyScale`:

    (gdb) bt
    #0  0x00007ffff6ac449e in QRasterPaintEngine::drawLines(QLine const*, int) ()
       from /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5
    #1  0x00007ffff6adc1cd in QPainter::drawLines(QLine const*, int) ()
       from /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5
    #2  0x000055555557e405 in SpectrogramPlot::paintFrequencyScale(QPainter&, QRect&) ()
    #3  0x0000555555575210 in PlotView::paintEvent(QPaintEvent*) ()
    #4  0x00007ffff7721038 in QWidget::event(QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5

schneider42, 2019-09-04:
I can only imagine that `bwPerTick` becomes a very small number, causing a huge number of loops

schneider42, 2019-09-04:
Actually...
```c
int tick = 0;

while (tick <= sampleRate / 2) {
```
I guess increasing the size of `tick` will be good enough.

miek, 2019-12-28:
This was due to high sample rates overflowing `size_t` & going negative, so I've switched to using `double` everywhere for sample rates (which also helps for future SigMF support).

commit d6115cb4 (2019-12-28): Use double for sample rate

Fixes #158
closed 2019-12-28 by miek