miek/inspectrum #160: Mac OSX - Homebrew & Macports CMakeLists.txt ERROR

DinBan, 2020-01-06:
Trying install it with Homebrew & Macports using instructions, but it not works, ERROR says it missing CMakeLists.txt when try to cmake ..

Here it is - 
1@MBP build % cmake ..
CMake Error: The source directory "/Users/1" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.

Any solution?
Thank you in advance!

domrim, 2020-01-14:
It looks like you not in the right directory. Looking at your error cmake searches in `/Users/1` the file `CMakeLists.txt`.
Let's assume you have cloned the source code to the dir `/Users/1/inspectrum`, so like described in the [wiki](https://github.com/miek/inspectrum/wiki/Build) you should create the folder `/Users/1/inspectrum/build` with the command `mkdir build` executed in `/Users/1/inspectrum`.
Then change your current folder to the newly created one with `cd build`.
Now `cmake ..` should run perfectly and also the rest of the compling-process should work fine.
Hope i could help :)

Note: hope you know, that you only need homebrew OR macports, not both ;)

Crane-Mocker, 2023-05-21:
you can use `brew install inspectrum`