miek/inspectrum #162: run inspectrum - xwindow appears and is totally empty black window

geetee24, 2020-02-12:
Any idea why empty window.  no UI?
[image]

sm0yyq, 2020-03-03:
It's the same for me on Ubuntu 19.10.

sm0yyq, 2020-03-08:
An update from me.

I have tried 3 distros, Ubuntu 19.10 and 18.04 and last Mint 18.
I have followed the build instructions, and i got all same results.
Nothing is showing in the windows after u loaded the file u want to use, it's completly blank as the example above.

I think this is a lite bad cose for example the good software "dspectrumui" also is dependent on Inspectrum, soo miek plz have a look at this.

domrim, 2020-03-08:
did you use the last release (v0.2.2) or have you cloned the git-repo and tried to build from master?

sm0yyq, 2020-03-08:
domrim: if it's me u asking i can tell u i used "git clone", but i can trie also to download the v0.2.2 source directly.

sm0yyq, 2020-03-08:
Well, can tell i have tried to download all sources from the beginning. All of whem is compiling well but same error on all of them. I cant get up anything to work on...the window is all clean :/

domrim, 2020-03-08:
@sm0yyq i quickly spawned an VM with Ubuntu 18.04 LTS adm64, then i run following commands:
```
sudo apt-get update -y
sudo apt-get install qt5-default libfftw3-dev cmake pkg-config libliquid-dev 
sudo apt-get install build-essential git
git clone https://github.com/miek/inspectrum.git
cd inspectrum/
mkdir build
cd build/
cmake ..
make
sudo make install
```
Now i can run `inspectrum` via the command line, and it works perfectly fine:
[image]

If your problem still persists, let me know what you did to compile & install inspectrum so we can find the issue.

sm0yyq, 2020-03-09:
Thx for ur answer domrim.

Ok lets see, i made a new fresh install of Ubuntu 18.04 LTS and i didn't upgrade it.

I follow ur steps as abow and it doesent work, i only got a black screen whatever file i try to load.
[image]

sm0yyq, 2020-03-09:
Well, i did install a Ubuntu 18.04 VM under Windows Virtualbox and it seems to work as for u.

So now i dont know why my "real" install of Ubuntu doesent work...???

May it be the graphic card or something.?

The system im using is a i7-930 and a Nvidia 760 GTX.

sm0yyq, 2020-03-09:
Well, a new update again.. i donwloaded this examples ..

https://www.sigidwiki.com/wiki/MOBITEX

And it seems to work in my "real" Ubuntu as well, .. Strange i cant use my other files ...

Do i need to "fipple" with the power sliders alot to accouly see anytning in the window?.

miek, 2020-11-23:
> Do i need to "fipple" with the power sliders alot to accouly see anytning in the window?.

Yep, though it really depends on the power level of signals in the file

closed 2020-11-23 by miek