miek/inspectrum #170: Vertical scrolling broken randomly after startup

schneider42, 2020-09-06:
Sometimes when I start Inspectrum, the vertical scrollbar doesn't show a slider:
[image]

After resizing the window, the slider appears:
[image]

I've put some logging into `plotview.cpp`:
```diff
@@ -181,6 +181,7 @@ void PlotView::enableCursors(bool enabled)
 }
 
 bool PlotView::viewportEvent(QEvent *event) {
+    qDebug() << "PlotView::viewportEvent " << event->type();
     // Handle wheel events for zooming (before the parent's handler to stop normal scrolling)
     if (event->type() == QEvent::Wheel) {
         QWheelEvent *wheelEvent = (QWheelEvent*)event;
```

Result when the slider appears after start:
```
PlotView::viewportEvent  QEvent::Type(CursorChange)
PlotView::viewportEvent  QEvent::Type(MouseTrackingChange)
PlotView::viewportEvent  QEvent::Type(Polish)
PlotView::viewportEvent  QEvent::Type(Move)
PlotView::viewportEvent  QEvent::Type(Resize)
PlotView::viewportEvent  QEvent::Type(Show)
PlotView::viewportEvent  QEvent::Type(ShowToParent)
PlotView::viewportEvent  QEvent::Type(PolishRequest)
PlotView::viewportEvent  QEvent::Type(Move)
PlotView::viewportEvent  QEvent::Type(Resize)
PlotView::viewportEvent  QEvent::Type(Move)
PlotView::viewportEvent  QEvent::Type(Resize)
PlotView::viewportEvent  QEvent::Type(UpdateLater)
PlotView::viewportEvent  QEvent::Type(Hide)
PlotView::viewportEvent  QEvent::Type(Move)
PlotView::viewportEvent  QEvent::Type(Resize)
PlotView::viewportEvent  QEvent::Type(Show)
PlotView::viewportEvent  QEvent::Type(Paint)
PlotView::viewportEvent  QEvent::Type(WindowActivate)
PlotView::viewportEvent  QEvent::Type(Move)
PlotView::viewportEvent  QEvent::Type(Resize)
PlotView::viewportEvent  QEvent::Type(Paint)
```

Result when the slider does not appear after start:
```
PlotView::viewportEvent  QEvent::Type(CursorChange)
PlotView::viewportEvent  QEvent::Type(MouseTrackingChange)
PlotView::viewportEvent  QEvent::Type(Polish)
PlotView::viewportEvent  QEvent::Type(Move)
PlotView::viewportEvent  QEvent::Type(Resize)
PlotView::viewportEvent  QEvent::Type(Show)
PlotView::viewportEvent  QEvent::Type(ShowToParent)
PlotView::viewportEvent  QEvent::Type(PolishRequest)
PlotView::viewportEvent  QEvent::Type(Move)
PlotView::viewportEvent  QEvent::Type(Resize)
PlotView::viewportEvent  QEvent::Type(Move)
PlotView::viewportEvent  QEvent::Type(Resize)
PlotView::viewportEvent  QEvent::Type(UpdateLater)
PlotView::viewportEvent  QEvent::Type(Hide)
PlotView::viewportEvent  QEvent::Type(Move)
PlotView::viewportEvent  QEvent::Type(Resize)
PlotView::viewportEvent  QEvent::Type(Move)
PlotView::viewportEvent  QEvent::Type(Resize)
PlotView::viewportEvent  QEvent::Type(Show)
PlotView::viewportEvent  QEvent::Type(Paint)
PlotView::viewportEvent  QEvent::Type(WindowActivate)
```

I'm not sure if the location of that print makes much sense, but there are differences between the two logs in terms of order.

My QT libraries have version `5.9.5` from Ubuntu 18.04

schneider42, 2020-09-06:
Some more debug prints:
```diff
@@ -181,6 +181,7 @@ void PlotView::enableCursors(bool enabled)
 }
 
 bool PlotView::viewportEvent(QEvent *event) {
+    qDebug() << "PlotView::viewportEvent " << event->type();
     // Handle wheel events for zooming (before the parent's handler to stop normal scrolling)
     if (event->type() == QEvent::Wheel) {
         QWheelEvent *wheelEvent = (QWheelEvent*)event;
@@ -369,6 +370,7 @@ void PlotView::invalidateEvent()
 
 void PlotView::repaint()
 {
+    qDebug() << "PlotView::repaint";
     viewport()->update();
 }
 
@@ -422,6 +424,7 @@ void PlotView::setPowerMax(int power)
 
 void PlotView::paintEvent(QPaintEvent *event)
 {
+    qDebug() << "PlotView::paintEvent " << event->type();
     if (mainSampleSource == nullptr) return;
 
     QRect rect = QRect(0, 0, width(), height());
@@ -517,6 +520,7 @@ int PlotView::plotsHeight()
 
 void PlotView::resizeEvent(QResizeEvent * event)
 {
+    qDebug() << "PlotView::resizeEvent " << event->type();
     updateView();
 }
 
@@ -548,6 +552,7 @@ void PlotView::updateViewRange(bool reCenter)
 
 void PlotView::updateView(bool reCenter)
 {
+    qDebug() << "PlotView::updateView";
     horizontalScrollBar()->setMaximum(std::max(0, sampleToColumn(mainSampleSource->count()) - width()));
     verticalScrollBar()->setMaximum(std::max(0, plotsHeight() - viewport()->height()));
     updateViewRange(reCenter);
```
Good:
```
PlotView::viewportEvent  QEvent::Type(CursorChange)
PlotView::viewportEvent  QEvent::Type(MouseTrackingChange)
PlotView::repaint
PlotView::repaint
PlotView::updateView
PlotView::repaint
PlotView::repaint
PlotView::repaint
PlotView::repaint
PlotView::repaint
PlotView::updateView
PlotView::repaint
PlotView::updateView
PlotView::updateView
PlotView::repaint
PlotView::repaint
PlotView::repaint
PlotView::viewportEvent  QEvent::Type(Polish)
PlotView::viewportEvent  QEvent::Type(Move)
PlotView::viewportEvent  QEvent::Type(Resize)
PlotView::resizeEvent  QEvent::Type(Resize)
PlotView::updateView
PlotView::viewportEvent  QEvent::Type(Show)
PlotView::viewportEvent  QEvent::Type(ShowToParent)
PlotView::viewportEvent  QEvent::Type(PolishRequest)
PlotView::viewportEvent  QEvent::Type(Move)
PlotView::viewportEvent  QEvent::Type(Resize)
PlotView::resizeEvent  QEvent::Type(Resize)
PlotView::updateView
PlotView::viewportEvent  QEvent::Type(Move)
PlotView::viewportEvent  QEvent::Type(Resize)
PlotView::resizeEvent  QEvent::Type(Resize)
PlotView::updateView
PlotView::viewportEvent  QEvent::Type(UpdateLater)
PlotView::viewportEvent  QEvent::Type(Hide)
PlotView::viewportEvent  QEvent::Type(Move)
PlotView::viewportEvent  QEvent::Type(Resize)
PlotView::resizeEvent  QEvent::Type(Resize)
PlotView::updateView
PlotView::viewportEvent  QEvent::Type(Show)
PlotView::viewportEvent  QEvent::Type(Paint)
PlotView::paintEvent  QEvent::Type(Paint)
PlotView::viewportEvent  QEvent::Type(Enter)
PlotView::viewportEvent  QEvent::Type(WindowActivate)
PlotView::viewportEvent  QEvent::Type(Move)
PlotView::viewportEvent  QEvent::Type(Resize)
PlotView::resizeEvent  QEvent::Type(Resize)
PlotView::updateView
PlotView::viewportEvent  QEvent::Type(Paint)
PlotView::paintEvent  QEvent::Type(Paint)
```

Bad:
```
PlotView::viewportEvent  QEvent::Type(CursorChange)
PlotView::viewportEvent  QEvent::Type(MouseTrackingChange)
PlotView::repaint
PlotView::repaint
PlotView::updateView
PlotView::repaint
PlotView::repaint
PlotView::repaint
PlotView::repaint
PlotView::repaint
PlotView::updateView
PlotView::repaint
PlotView::updateView
PlotView::updateView
PlotView::repaint
PlotView::repaint
PlotView::repaint
PlotView::viewportEvent  QEvent::Type(Polish)
PlotView::viewportEvent  QEvent::Type(Move)
PlotView::viewportEvent  QEvent::Type(Resize)
PlotView::resizeEvent  QEvent::Type(Resize)
PlotView::updateView
PlotView::viewportEvent  QEvent::Type(Show)
PlotView::viewportEvent  QEvent::Type(ShowToParent)
PlotView::viewportEvent  QEvent::Type(PolishRequest)
PlotView::viewportEvent  QEvent::Type(Move)
PlotView::viewportEvent  QEvent::Type(Resize)
PlotView::resizeEvent  QEvent::Type(Resize)
PlotView::updateView
PlotView::viewportEvent  QEvent::Type(Move)
PlotView::viewportEvent  QEvent::Type(Resize)
PlotView::resizeEvent  QEvent::Type(Resize)
PlotView::updateView
PlotView::viewportEvent  QEvent::Type(UpdateLater)
PlotView::viewportEvent  QEvent::Type(Hide)
PlotView::viewportEvent  QEvent::Type(Move)
PlotView::viewportEvent  QEvent::Type(Resize)
PlotView::resizeEvent  QEvent::Type(Resize)
PlotView::updateView
PlotView::viewportEvent  QEvent::Type(Move)
PlotView::viewportEvent  QEvent::Type(Resize)
PlotView::resizeEvent  QEvent::Type(Resize)
PlotView::updateView
PlotView::viewportEvent  QEvent::Type(Show)
PlotView::viewportEvent  QEvent::Type(Paint)
PlotView::paintEvent  QEvent::Type(Paint)
PlotView::viewportEvent  QEvent::Type(WindowActivate)
```

miek, 2020-09-06:
Just to make sure I'm understanding right, I think you're actually talking about the horizontal scrollbar? (ie: for scrolling left or right in time).

So what seems to be happening is that the range of the scrollbar isn't getting set correctly. It has a slider, but it's full width because there's no range of movement available.
We set up the slider range in `updateView` based on the sample count and zoom/fft levels. It seems like in the 'bad' log, there's no `updateView` call after the `WindowActivate` event, so maybe the sliders end up just having default settings until some action triggers `updateView` to fix it.

schneider42, 2020-09-06:
You are of course correct: I meant the horizontal scrollbar...

schneider42, 2020-09-06:
This patch solved this issue for me. It feels like a hack, though:
```patch
@@ -237,6 +238,10 @@ bool PlotView::viewportEvent(QEvent *event) {
                 return true;
     }
 
+    if (event->type() == QEvent::WindowActivate) {
+        updateView();
+    }
+
     // Handle parent eveents
     return QGraphicsView::viewportEvent(event);
 }
```

schneider42, 2020-11-23:
Closed by #177

closed 2020-11-23 by schneider42