miek/inspectrum #173: no horizontal slider after startup

geo207, 2020-10-03:
Hello,

I downloaded inspectrum a few days ago, but there is no horizontal zoom slider on the signal area (see first screenshot). I tried resizing and adjusting the window as suggested in issue #170, but it had no effect.  Also, when I right click and then click on one of the four options, a new area opens at the bottom of the window, and if I do that again then another new area opens at the bottom; this time the second area is a bright green, but with other signal files it is a different color (see second screenshot).  I don't think these new areas are supposed to appear based on what I've read about inspectrum online.  I am using a Raspberry Pi 4 (8 GB) and the version of inspectrum is 0.2-1.  In the terminal window, when I start inspectrum there are always 6 lines that read, "libpng warning: iCCP: known incorrect sRGB profile"   I don't know if this is connected to the horizontal zoom slider missing.  Can anyone tell me why I am not getting the slider and what the remedy might be?  I don't understand the coding solution in #170 and I'm not sure it pertains to my problem in any case.  I'm very disappointed as I have signal files ready to roll!  Thanks and here are two screenshots:  

[image]

[image]

geo207, 2020-10-04:
twelve hours later....I think I figured it out now.  Still working out how to count symbols, but I think I found the cursor and the zoom. 

 
[image]

miek, 2020-11-23:
> Still working out how to count symbols

You can drag the cursors so that a transition in the signal always lands on a dotted line, and that will setup the symbol rate roughly. Then you can increase the symbol count and the cursors will expand out to cover more of the signal & count out the symbols (you may need to continue tweaking the rate as you go)

closed 2020-11-23 by miek