miek/inspectrum #181: Submodule libsigmf?

miek, 2021-02-27:
It might be a good idea to submodule libsigmf rather than making users (& package maintainers) install it for SigMF support

schneider42, 2021-03-01:
@Sec42 recently had quite a hard time installing libsigmf into his (pybombs) installation. I think he might have some input here.

Sec42, 2021-03-02:
I'm not sure if it's really applicable to this question of using a submoduleor not.

I had trouble building it in a pybombs environment. To get it to build cleanly I had to use specific (old) versions of flatbuffers & nlohmann_json.

source: git+https://github.com/google/flatbuffers
gitrev: 71aca81ff9

and

source: git+https://github.com/nlohmann/json.git
gitbranch: v3.7.1

closed 2022-03-26 by miek