miek/inspectrum #185: debian  cmake build errors

auskimus, 2021-04-14:
Hi

when I run cmake ..
I get the followin any help.

-- The CXX compiler identification is GNU 9.3.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Found FFTW: /usr/lib/x86_64-linux-gnu/libfftw3f.so  
-- Found LIQUID: /usr/local/lib/libliquid.so  
CMake Error at /usr/share/cmake-3.16/Modules/CMakeFindDependencyMacro.cmake:47 (find_package):
  By not providing "Findnlohmann_json.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "nlohmann_json", but CMake did not find one.

  Could not find a package configuration file provided by "nlohmann_json"
  with any of the following names:

    nlohmann_jsonConfig.cmake
    nlohmann_json-config.cmake

  Add the installation prefix of "nlohmann_json" to CMAKE_PREFIX_PATH or set
  "nlohmann_json_DIR" to a directory containing one of the above files.  If
  "nlohmann_json" provides a separate development package or SDK, be sure it
  has been installed.
Call Stack (most recent call first):
  /usr/local/lib/cmake/libsigmf/libsigmfConfig.cmake:15 (find_dependency)
  src/CMakeLists.txt:59 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/auski/inspectrum/build/CMakeFiles/CMakeOutput.log".

 Thanks in advance

miek, 2021-04-14:
Try installing the `nlohmann-json3-dev` package

auskimus, 2021-04-14:
Hi 

I did install nlohmann-json3-dev package, but still got the following error;

CMake Error at /usr/share/cmake-3.16/Modules/CMakeFindDependencyMacro.cmake:47 (find_package):
  By not providing "FindFlatbuffers.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "Flatbuffers", but CMake did not find one.

  Could not find a package configuration file provided by "Flatbuffers" with
  any of the following names:

    FlatbuffersConfig.cmake
    flatbuffers-config.cmake

  Add the installation prefix of "Flatbuffers" to CMAKE_PREFIX_PATH or set
  "Flatbuffers_DIR" to a directory containing one of the above files.  If
  "Flatbuffers" provides a separate development package or SDK, be sure it
  has been installed.
Call Stack (most recent call first):
  /usr/local/lib/cmake/libsigmf/libsigmfConfig.cmake:16 (find_dependency)
  src/CMakeLists.txt:59 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/auski/inspectrum/build/CMakeFiles/CMakeOutput.log".

miek, 2021-04-14:
You probably need the `libflatbuffers-dev` package too

auskimus, 2021-04-14:
Thank you very much

installation completed, but with black screen, is it graphic drivers or i might be missing something? I am on Dell XPS Ubuntu 20.04.

auskimus, 2021-04-14:
[image]

auskimus, 2021-04-14:
that image from the net but resemble what I get

miek, 2021-04-14:
You may just need to adjust the power min/max sliders for a good image. Also the file-type in that image is unsupported, but that may not apply to you (there's a full list of types here: https://github.com/miek/inspectrum#input)

auskimus, 2021-04-14:
Thanks a lot Miek, will close now

closed 2021-04-14 by auskimus