miek/inspectrum #209: Reading SDRangel SigMF files

PeteK6TJ, 2022-07-17:
SDRangel seems to write ci32 SigMF files, which it seems neither inspectrum nor gnuradio can read as SigMF.  But in the inputsource.cpp source file of inspectrum, I added a ci32_le type entry and created a S32 sample converter (and a few others) by straightforward plagiarizing of the sample converters already in there.  Now I can snoop around for short packets with SDRangel (which has a fast display), save a giant file, and use inspectrum to find what I want to analyze in gnuradio.  Open and closed feature request??

catkira, 2022-10-17:
I would be interested

PeteK6TJ, 2022-10-17:
Does this email work??

Pete Mahowald | ***@***.*** | 650 492-3989

catkira, 2022-10-18:
You can reach me at benjamin@menkuec.de

PeteK6TJ, 2022-10-18:
> I quickly did a PR that adds support for ci32, since there is no answer
> here.
> #213 <https://github.com/miek/inspectrum/pull/213>

Here's my code.

catkira, 2022-10-18:
I dont see any code, but I already re-created the PR see here: https://github.com/miek/inspectrum/pull/213

schneider42, 2022-10-19:
This has been merged. Once you build/install inspectrum from source you will be able to make use of the feature. I don't know about timelines for a release so distributions can pick up the change.

catkira, 2022-10-19:
Yes I know, I made the post before it was merged ^^