miek/inspectrum #219: Zero pad to compensate for nonzero FFT length

jacobagilbert, 2022-11-21:
Instantaneous energy and the spectrogram tiles do not line up correctly in Inspectrum. This is most obvious when using larger FFT lengths and the amplitude plot, but also very noticeable with SigMF Annotations, for example, in the screenshot below shows the spectrogram tile showing energy well before the annotation and actual time domain energy begins:

[image]

At larger FFT sizes this misalignment becomes more obvious as the temporal ambiguity (same number of samples as FFT length) increases.

To resolve this, the data should be zero padded with length `NFFT/2` in which case the spectrogram will be nicely aligned (with expected temporal smearing still of course):

[image]

daniestevez, 2023-10-10:
Just seen this issue. I've also realized this problem, and sent a PR that solves the problem in a slightly different way (#227).

miek, 2023-10-22:
Fixed in #227

closed 2023-10-22 by miek