miek/inspectrum #239: Crashes on addition of derived plot

maxandroverforever, 2025-01-10:
Whenever I attempt to create a derived plot, the program crashes. No errors were outputted to the console. I'm using radioconda on Windows 11.

BlazCussigh, 2025-01-23:
> Whenever I attempt to create a derived plot, the program crashes. No errors were outputted to the console. I'm using radioconda on Windows 11.

Did you fix it?

maxandroverforever, 2025-01-23:
I have not fixed it, I just ran it in an Ubuntu VM, which works fine.

BlazCussigh, 2025-01-23:
I also have Ubuntu WM but when i try to extract bits from the signal it doesn’t give me 0 and 1 but instead gives some random number some are negative and some pozitive. Do you maybe know what i did wrong?

maxandroverforever, 2025-01-24:
Do you have a threshold plot of the frequency plot with the threshold set correctly?

BlazCussigh, 2025-01-24:
Can I provide you a picture of my page as I’m kinda new in this field and I have to make a project for my collage where I extract bits from signala. I would really appreciate a bit of help.

maxandroverforever, 2025-01-28:
Sure, what is it you're trying to do exactly?

AgustinR441, 2025-05-09:
Same problem here

ckoval7, 2025-12-06:
I am having this issue as well.